nasa/nos3 #602: 🐛 [BUG] - <title>1.7.2 stop

gardinwin, 2025-04-02:
### Description

When version 1.7.2 is installed and run on Linux, it stops at almost the same point in time as shown in the referenced screen.
How can this issue be resolved?

### Reproduction steps

```bash
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
```

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

### Logs

```bash

```

### OS

Linux (Direct Clone)

gardinwin, 2025-04-02:
[https://drive.google.com/file/d/120Ts54oYpFPLAmeceC1ItricWBmnm_A1/view?usp=sharing](url)

gardinwin, 2025-04-02:
https://drive.google.com/file/d/120Ts54oYpFPLAmeceC1ItricWBmnm_A1/view?usp=sharing

gardinwin, 2025-04-02:
https://drive.google.com/file/d/1tQxRjwV3MBzmMtKje8nLFWn9zOlWeB2n/view?usp=sharing

gardinwin, 2025-04-02:
error logs :
OS_BSP_Initialize():Maximum user msg queue depth = 10000
   *                                      *                    *
       *                                                  *                *             
            ]]]]]]]]]]]]  ]]]]]]]]]]]]]]]]]  ]]]]]]]]]]]]]]]]            ]]]]]]
          ]]]]]]]]]]]]]  ]]]]]]]]]]]]]]]]]  ]]]]]]]]]]]]]]]]            ]]]]]] 
    *    ]]]]]]]]]]]]]  ]]]]]]]]]]]]]]]]]  ]]]]]]]]]]]]]]]]            ]]]]]]  
         ]]]]]]              ]]]]]]       ]]]]]]]                     ]]]]]]   
         ]]]]]]]]       *   ]]]]]]       ]]]]]]]]]]]]]]   ]]]]]]]]   ]]]]]]    
 *        ]]]]]]]]         ]]]]]]       ]]]]]]]]]]]]]]   ]]]]]]]]   ]]]]]]     
             ]]]]]]      ]]]]]]]  *    ]]]]]]]                     ]]]]]]      
   ]]]]]]]]]]]]]]]      ]]]]]]]       ]]]]]]]   FLIGHT SOFTWARE   ]]]]]]     *  
  ]]]]]]]]]]]]]]]      ]]]]]]]       ]]]]]]]                     ]]]]]]        
 ]]]]]]]]]]]]]]       ]]]]]]]       ]]]]]]]                     ]]]]]]
      *                        *                        *                         *
 *                                     *                                *               *
                *

CFE_PSP: Reset Type: PO
CFE_PSP: Default Reset SubType = 1
CFE_PSP: Default CPU ID = 1
CFE_PSP: Default Spacecraft ID = 42
CFE_PSP: Default CPU Name: cpu1
OS_Posix_GetSchedulerParams():191:Policy 1: available, min-max: 1-99
OS_Posix_GetSchedulerParams():191:Policy 2: available, min-max: 1-99
OS_Posix_TaskAPI_Impl_Init():377:Selected policy 2 for RT tasks, root task = 99
initializing nos engine link...
CFE_PSP: initializing module 'eeprom_notimpl' with ID 0110ff00
CFE_PSP: EEPROM access not implemented
CFE_PSP: initializing module 'port_notimpl' with ID 0110ff01
CFE_PSP: I/O Port access not implemented
CFE_PSP: initializing module 'ram_notimpl' with ID 0110ff02
CFE_PSP: Physical RAM access not implemented
CFE_PSP: Starting the cFE with a POWER ON reset.
CFE_PSP: Clearing out CFE CDS Shared memory segment.
CFE_PSP: Clearing out CFE Reset Shared memory segment.
CFE_PSP: Clearing out CFE User Reserved Shared memory segment.
2000-001-00:00:00.00000 CFE_ES_SetupResetVariables: POWER ON RESET due to Power Cycle (Power Cycle).
2000-001-00:00:00.00000 CFE_ES_Main: CFE_ES_Main in EARLY_INIT state
CFE_PSP: CFE_PSP_AttachExceptions Called
2000-001-00:00:00.00000 CFE_ES_Main: CFE_ES_Main entering CORE_STARTUP state
2000-001-00:00:00.00000 CFE_ES_CreateObjects: Starting Object Creation calls.
2000-001-00:00:00.00000 CFE_ES_CreateObjects: Calling CFE_Config_Init
2000-001-00:00:00.00000 CFE_ES_CreateObjects: Calling CFE_ES_CDSEarlyInit
2000-001-00:00:00.00000 CFE_ES_CreateObjects: Calling CFE_EVS_EarlyInit
2000-001-00:00:00.00000 CFE_EVS_EarlyInit: Event Log cleared following power-on reset
2000-001-00:00:00.00000 CFE_ES_CreateObjects: Calling CFE_SB_EarlyInit
2000-001-00:00:00.00000 CFE_ES_CreateObjects: Calling CFE_TIME_EarlyInit
2000-001-00:00:00.00000 CFE_ES_CreateObjects: Calling CFE_TBL_EarlyInit
2000-001-00:00:00.00000 CFE_ES_CreateObjects: Calling CFE_FS_EarlyInit
EVS Port1 EVS Port1 42/1/CFE_EVS 1: cFE EVS Initialized:  cFE DEVELOPMENT BUILD v7.0.0-rc4+dev370 (Codename: Draco), Last Official Release: cfe v6.7.0
2000-001-00:00:00.15000 CFE_SB_AppInit: Registered 4 events for filtering
EVS Port1 EVS Port1 42/1/CFE_SB 1: cFE SB Initialized:  cFE DEVELOPMENT BUILD v7.0.0-rc4+dev370 (Codename: Draco), Last Official Release: cfe v6.7.0
EVS Port1 EVS Port1 42/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_SB
EVS Port1 EVS Port1 42/1/CFE_ES 1: cFE ES Initialized:  cFE DEVELOPMENT BUILD v7.0.0-rc4+dev370 (Codename: Draco), Last Official Release: cfe v6.7.0
EVS Port1 EVS Port1 42/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_ES
EVS Port1 EVS Port1 42/1/CFE_ES 2: cFS Versions: cfe v7.0.0-rc4+dev370, osal v6.0.0-rc4+dev223, psp v1.6.0-rc4+dev67. cFE chksm 40069
EVS Port1 EVS Port1 42/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_ES
EVS Port1 EVS Port1 42/1/CFE_ES 91: Version Info: Mission NOS3, version git:v1_07_02-dirty
EVS Port1 EVS Port1 42/1/CFE_SB 14: No subscribers for MsgId 0x808,sender CFE_ES
EVS Port1 EVS Port1 42/1/CFE_ES 91: Version Info: Module CORE_API, version git:draco-rc5-7-g87e27374
EVS Port1 EVS Port1 42/1/CFE_ES 91: Version Info: Module ES, version git:draco-rc5-7-g87e27374
EVS Port1 EVS Port1 42/1/CFE_ES 91: Version Info: Module EVS, version git:draco-rc5-7-g87e27374
EVS Port1 EVS Port1 42/1/CFE_ES 91: Version Info: Module FS, version git:draco-rc5-7-g87e27374
EVS Port1 EVS Port1 42/1/CFE_ES 91: Version Info: Module SB, version git:draco-rc5-7-g87e27374
EVS Port1 EVS Port1 42/1/CFE_ES 91: Version Info: Module TBL, version git:draco-rc5-7-g87e27374
EVS Port1 EVS Port1 42/1/CFE_ES 91: Version Info: Module TIME, version git:draco-rc5-7-g87e27374
EVS Port1 EVS Port1 42/1/CFE_ES 91: Version Info: Module OSAL, version git:draco-rc5-23-g08a79bb6
EVS Port1 EVS Port1 42/1/CFE_ES 91: Version Info: Module PSP, version git:equuleus-rc1-30-g320bc95
EVS Port1 EVS Port1 42/1/CFE_ES 91: Version Info: Module MSG, version git:draco-rc5-7-g87e27374
[... 152 lines cut ...]
EVS Port1 EVS Port1 42/1/CFE_TIME 21: Stop FLYWHEEL
EVS Port1 EVS Port1 42/1/SBN 8: Processing all subscriptions...
EVS Port1 EVS Port1 42/1/SBN 515: connecting to peer (PeerData=0x7fc35571cfc4, ProcessorID=2)
EVS Port1 EVS Port1 42/1/SBN 513: unable to connect to peer (PeerData=0x7fc35571cfc4)
[... cut ...]
EVS Port1 EVS Port1 42/1/CSS 14: GENERIC_CSS: Device enabled
EVS Port1 EVS Port1 42/1/FSS 13: GENERIC_FSS: Enable command received
EVS Port1 EVS Port1 42/1/FSS 14: GENERIC_FSS: Device enabled
EVS Port1 EVS Port1 42/1/SBN 5: Created peer pipe 'SBN_2_42_Pipe', ID '1441831'
EVS Port1 EVS Port1 42/1/SBN 5: Peer 42:2 connected.
EVS Port1 EVS Port1 42/1/FSS 31: GENERIC_FSS: Request device data reported error -1
snb_subs: ProcessSubFromPeer: MyProcessorID = 1, MySpacecraftID = 42, PeerProcessorID = 2, PeerSpacecraftID = 42, MsgID = 0x08fa

jlucas9, 2025-04-02:
Hi @gardinwin,

Just wanted to confirm that you have resolved this issue. What was your mitigation if so in case someone else finds this issue?

Thanks!

gardinwin, 2025-04-02:
not solved
sometimes nos stopped
please help me

감사합니다.
강병국 배상

==================================================
(주)폴모션
상무이사 / 수석연구원

강병국
P: 010-6859-6062
M: ***@***.***,        ***@***.***

jlucas9, 2025-04-02:
Let me see if I can duplicate this. 

I'm running on the Vagrant VM included with version 1.7.2 (nos3_20250217), note this specific VM does not have anything special installed apart from Docker, Python, and VSCode. 

In the terminal:
```
cd ~/Downloads
git clone https://github.com/nasa/nos3.git
cd nos3
git submodule update --init --recursive
```

[image]

In the terminal:
```
make prep
```

[image]

At this point you may want to reconfigure to use COSMOS instead of the new default ground software YAMCS.
To do this, edit the `./cfg/nos3-mission.xml` line 7 to be `<gsw>cosmos</gsw>`.
I'll do this here as well.

[image]

In the terminal:
```
make
make launch
```

[image]

Rearranging some windows.

[image]

It doesn't appear I'm able to reproduce directly base don the above and we eventually get to a sun pointed spacecraft and time continues to move.

[image]

----------------------------

Looking at your above logs it seems that maybe FSS or another component is not connecting to 42 or is perhaps crashing out. 

Can you confirm that all expected simulators are still running? The gnome-terminal drop down might be easiest to compare.

[image]

Also can you do a `git reset --hard --recurse-submodules` and also confirm your `git status` up to date?

[image]

jlucas9, 2025-04-02:
I may have just been able to recreate this on my standard development VM on the latest `dev` branch [bbd58a3]. I didn't get it in the screenshot but possibly the FSS simulator crashed.

[image]

This wasn't repeatable and after the shown `make stop` the following `make launch` immediately seemed to resolve it.

[image]

gardinwin, 2025-04-02:
Thank you for your response.
I am running it directly on Linux instead of using Vagrant to improve performance.
Sometimes it runs fine, but most of the time it seems to run and then halts.
I am using Ubuntu 20.04 LTS.
Thank you.

gardinwin, 2025-04-07:
Hi
Most of the issues seem to occur at the point when 42 starts.
The problem occurs when connecting to 42, regardless of whether it is in GUI or non-GUI mode.

Thank you

closed 2025-08-28 by zlynch2