nasa/nos3 #610: 🐛 [BUG] - <title> 1.7.2 work wrong ?

gardinwin, 2025-04-07:
### Description

The FSS component does not update its telemetry regardless of the commands sent.
Is there another command required to enable it?

The ARDUCAM also does not update its telemetry after the HW_CHECK command.

### Branch Name

main

### Reproduction steps

```bash
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
```

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

### Logs

```bash

```

### OS

Linux (Direct Clone)

gardinwin, 2025-04-07:
1.6.x works well.

gardinwin, 2025-04-08:
why 42 console print out "Imsg limit exceeded" messages ?

jlucas9, 2025-04-10:
Hi @gardinwin,

The `Imsg limit exceeded` message from 42 doesn't seem to cause any issues for us but be a bug causing the spam messages within the 42 code. When you boot up can you try to disable the FSS in the spacecraft configuration file?

Change https://github.com/nasa/nos3/blob/main/cfg/sc-full-config.xml#L37 to false, do a `make clean` and then `make` and see if that fixes anything for you.

gardinwin, 2025-04-14:
That part doesn’t seem to be related.
If 42 is run independently, the message does not appear.
It seems there is an issue with the part where 42 interacts with the SIM.