nasa/nos3 #627: Scenario - STF add FDC to check for sample disabled while in science

jlucas9, 2025-04-10:
### Summary

Develop a scenario that showcases the sample component disabling itself when an error in the simulator was detected, then capture the steps to perform updates to the LC and SC tables as needed to automatically power cycle it if this occurs while in a science mode.

### Use Case

Demonstrate some additional logic being added to an existing mission configuration.

dccutrig, 2025-04-28:
@zlynch2 I don't want to overcomplicate this but was thinking along the lines of the following:

- Add a 'SelfTest' or 'BuiltInTest' HK Field that defaults to pass on every bootup
- Make it 'Fail' via SimBackDoor command, simulating an internal fault
-  'Fail' should be relayed to the ground in HK

^ Everything above should already exist for this scenario

On the ground, the operator needs to:
- Define a new watchpoint to monitor the SelfTest HK Field
- Define a new actionpoint to monitor the SelfTest Watchpoint
- Write a new RTS that will need to do the following:
-     Disable WPs, or Disable LC Entirely
-     Call pre-existing exit science RTS (I think off the top of my head)
-     Determine if we should reboot into science mode, to see if fault clears, or into safe mode (Option B seems more likely, A could cause reboot loop and dangerous... but we could toy with the idea that the fault doesn't necessarily clear on reboot and they brick a spacecraft)
-     Upload new WP table, AP table, Upload new RTS, Update existing RTSs that need reference to the new Watchpoint (when to reset and activate while changing between science / safe nominally)

zlynch2, 2025-04-29:
I can start to walk through this with screen shots if you havent already. I think we need to call out the file locations and whats changing here step by step, then provide a list for what to watch for like above.

jlucas9, 2025-05-06:
Merged into `dev` ^

closed 2025-05-06 by jlucas9