nasa/nos3 #644: Time Configuration

jlucas9, 2025-04-16:
### Description

The time should be set in the spacecraft configuration file instead of mission and also propagate to the various pieces that need it. 

### Reproduction steps

```bash
...
```

### Screenshots

```bash
=
```

### Logs

```bash

```

### OS

Windows (Direct Clone)

dccutrig, 2025-04-17:
As an aside, Robert ran into something similar with this. There may be a good reason to set the time in NOS, propagate it to 42 so sensors, 42, and NOS are in sync, and then manage spacecraft time as part of an optional timejam operations routine. (Default should probably be everything syncs on boot up though.)

Issues / questions relating to the different time formats and conversions between them to propagate into the different files.

jlucas9, 2025-05-07:
Merged into `dev` ^