nasa/nos3 #684: Scenario - Flight Build

jlucas9, 2025-05-07:
### Summary

Create a walkthrough scenario to walk through adding a flight target. 

Provide an example of updating the docker file with the desired toolchain, checking to use that new dockerfile, adding in the cFS target, and then show it building.

### Use Case

Missions who leverage NOS3 will want to know how to take it to the next level and build for there flight target.

jechiasson, 2025-05-08:
Thanks.  I'm trying this right now and failing to get cross compiling working.  Maybe an example with a common target (qemu or Raspberry Pi?) would be illustrative.

jlucas9, 2025-05-14:
@jechiasson, that's the one! I wrote up some information for the readme in this PR if you wanted to take a look and let me know if anything doesn't make sense - https://github.com/nasa/nos3/pull/693
Thanks!

jlucas9, 2025-05-23:
Merged into `dev`

closed 2025-05-23 by jlucas9

jechiasson, 2025-05-30:
Thanks @jlucas9. I'm building successfully, but am not sure what I should expect on launch.  Should the fsw container respond to cosmos similarly to the nos3 build?

jechiasson, 2025-06-09:
When cross-compiling to arm64 using toolchain-arm64-posix.cmake, and these settings:

SET(CFE_SYSTEM_PSPNAME      "pc-linux")
SET(OSAL_SYSTEM_OSTYPE      "posix")
SET(OSAL_SYSTEM_BSPTYPE     "generic-linux")

the OSAL is using it's version of default_config.cmake.  What else needs to be set to use the default_config.cmake in nos3_defs?