nasa/nos3 #695: 🐛 [BUG] - <title> linking on RISC-V 64 fails

ex0atmospheric, 2025-05-14:
### Description

environment is debian on qemu riscv64

running the make command in the fsw directory compiles the objects but the linker fails:

[ 97%] Building C object cpu1/CMakeFiles/core-cpu1.dir/home/ex0atmospheric/cFS/build/src/cfe_mission_strings.c.o
[ 98%] Building C object cpu1/CMakeFiles/core-cpu1.dir/cfe_module_version_table.c.o
[ 98%] Building C object cpu1/CMakeFiles/core-cpu1.dir/cfe_build_env_table.c.o
[ 98%] Building C object cpu1/CMakeFiles/core-cpu1.dir/cfe_psp_module_list.c.o
[ 98%] Building C object cpu1/CMakeFiles/core-cpu1.dir/cfe_static_symbol_list.c.o
[100%] Building C object cpu1/CMakeFiles/core-cpu1.dir/cfe_static_module_list.c.o
[100%] Building C object cpu1/CMakeFiles/core-cpu1.dir/src/target_config.c.o
[100%] Linking C executable core-cpu1
/usr/bin/ld: ../psp/libpsp-pc-linux.a(cfe_psp_memory.c.o): in function `CFE_PSP_InitProcessorReservedMemory':
/home/ex0atmospheric/cFS/psp/fsw/pc-linux/src/cfe_psp_memory.c:644:(.text+0x95a): undefined reference to `_init'
/usr/bin/ld: /home/ex0atmospheric/cFS/psp/fsw/pc-linux/src/cfe_psp_memory.c:646:(.text+0x968): undefined reference to `_fini'
/usr/bin/ld: /home/ex0atmospheric/cFS/psp/fsw/pc-linux/src/cfe_psp_memory.c:646:(.text+0x970): undefined reference to `_init'
collect2: error: ld returned 1 exit status
make[7]: *** [cpu1/CMakeFiles/core-cpu1.dir/build.make:225: cpu1/core-cpu1] Error 1
make[6]: *** [CMakeFiles/Makefile2:2529: cpu1/CMakeFiles/core-cpu1.dir/all] Error 2
make[5]: *** [Makefile:146: all] Error 2
make[4]: *** [CMakeFiles/native_default_cpu1-all.dir/build.make:70: CMakeFiles/native_default_cpu1-all] Error 2
make[3]: *** [CMakeFiles/Makefile2:631: CMakeFiles/native_default_cpu1-all.dir/all] Error 2
make[2]: *** [CMakeFiles/Makefile2:270: CMakeFiles/mission-all.dir/rule] Error 2
make[1]: *** [Makefile:179: mission-all] Error 2
make: *** [Makefile:123: all] Error 2
ex0atmospheric@riscv64 ~/cFS % 

### Reproduction steps

```bash
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
```

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

### Logs

```bash

```

### OS

Linux (Direct Clone)

kevincbruce, 2025-05-23:
I don't believe we have tested or verified in RISC-V 64 of yet, so likely our precompiled packages are not working properly, or there is some incompatible library. For now, we have been primarily using x86-64/amd64, with some attention to ARM support. You definitely can adapt things to cross compile for different architectures and toolchains, as shown in our guide here: https://nos3.readthedocs.io/en/latest/Scenario_Flight_Build.html; however, it generally takes a lot of tweaking, and NOS3 itself is only fully tested on AMD64/x86-64, and partially tested/compatible with arm64.

I hope that helps.

ex0atmospheric, 2025-05-25:
I know from my time at Northrop Grumman, that SPARC is highly used.I knit from my time at Terran Orbital that ARM is highly used.RISC-V is fast rising due to processors/cores.Dave

kevincbruce, 2025-05-25:
Yes, we have crosscompiled for SPARC boards on projects before, if we have the right tool chain. It works for flight software. But at the moment, parts of the simulator suite are reliant on packages that we have only fully compiled, tested, and set up for x86-64, AMD64, and somewhat ARM64 hosts.

ex0atmospheric, 2025-05-27:
> Hi @ex0atmospheric,
> The simulators for NOS3 don't actually run on the flight target is the difference I think here. CPU1 in cFS is setup to be nos-linux so that we can control time and link the middleware to the simulators, on real flight hardware you would use the real interface (sockets/CAN/I2C/SPI/GPIO) and connect to the real component on the other side. I'd recommend you keep CPU1 setup for NOS3 / nos-linux then add a CPU2 target for your flight build.
> Also a small note about the Scenario - Flight Build, particularly the part where we remove a large number of targets as I didn't want to go through the process detailing how to build a custom toolchain that includes the additional required packages such a libsocketcan-dev in your toolchain required by HWLIB and flight board Linux deployment as that could be a whole class in itself.

I was attempting to compile on a QEMU RISC-V Debian environment.As a side query, if you build for ARM Debian target, does just the Core CPU executable taken over or are there additional support files required that are cFS/cFE related?Dave

closed 2025-08-26 by zlynch2