nasa/nos3 #708: 🐛 [BUG] - OpenC3 CFS_RADIO TO_ENABLE_OUTPUT not working with default configuration

alexxhoss, 2025-05-22:
### Description

Is openc3 still supported out of the box?

I've been able to follow the demo https://nos3.readthedocs.io/en/latest/Scenario_Demo.html with COSMOS4 with much success, but when switching to use openc3 (and of course cleaning up the repo with 'make clean'/'git submodules...'/make' etc), I simply cannot get CFS_RADIO to work.

This is consistent in a few different environments that I've tried and I've just now defaulted back to the vagrant setup on the latest version of _main_ to remove all variables and have found that utilizing CFS_RADIO, I cannot get TLM packets to start sending across the RADIO interface , nor can I see any output from the NOS3 terminal to verify that any command is being received.

I have also tried different branches and versions of the code base, like several different versions of dev (including the commit 95488ce as seen in the link above), and tag v1_07_01 of main.

Additionally, `scripts/gsw/stop_gsw.sh` file seems to be out of date and the path of the env.sh file is wrong and the line should be updated to the following:
```
source $SCRIPT_DIR/../env.sh
```
which has lead me to believe openc3 is likely not supported/tested regularly. 

What can I do to troubleshoot and resolve the problem to properly utilize the CFS_RADIO component?  Note that running the same command through CFS_DEBUG _does_ work, but ideally I'm trying to learn and utilize the software the "right" way.

Thanks for any assistance, I have plenty more info to give as needed!

[image]

### Branch Name

main

### Reproduction steps

```bash
1. `git submodule update --init --recursive`
2. edit 'cfg/nos3-mission.xml' to use 'openc3'
3. `make prep`
4. `make`
5. Open Firefox to localhost:2900 & ensure openc3-cosmos-init container has exited 0
6. `make launch`
7. Attempt to send 'CFS_RADIO TO_ENABLE_OUTPUT' packet with default configuration
```

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

### Logs

```bash

```

### OS

Linux (Direct Clone)

kevincbruce, 2025-05-23:
Hello,

I went to review, switched config to OpenC3, and went to the newest Dev Branch, and found TO_ENABLE_OUTPUT worked for me. I sent it (over debug, and started receiving data over the Radio channel, as I would expect, and as illustrated below. Can try to look at it more if there's more specifics about reproduction.

[image]

alexxhoss, 2025-05-27:
Hey!

My apologies, I wasn't being very clear. The short story is that in default state version of _main_ is that the RADIO interface isn't working to send commands in the OpenC3 UI, or at least the basic commands that I've tested are not working.

My steps:

1. `git checkout main && git pull`
2. `git submodule update --init --recursive` and ensure all submodules are properly up to date / git repo is clean
3. edit 'cfg/nos3-mission.xml' to use 'openc3'
4. `make prep`
5. `make`
6. Open Firefox to localhost:2900 & ensure openc3-cosmos-init container has exited 0
7. `make launch`
8. Go to Command Sender, attempt to send 'CFS_RADIO CFE_ES_NOOP' command

At this point, when working, I would expect to see output in the NOS3 terminal to see that it registered my command... which is not currently happening.

After continuing to dig, it seems that this might be related to cryptolib.  I have traced back to a commit from a little over a year ago where functionality is working as expected, and then I've identified the commit that I believe broke the functionality.