nasa/nos3 #709: 🐛 - No Launch

byt3n33dl3, 2025-05-23:
### Description

sudo apt install binfmt

```docker
docker run --rm --privileged tonistiigi/binfmt --install all

docker run --rm --privileged tonistiigi/binfmt --uninstall qemu-x86_64
docker run --rm --privileged tonistiigi/binfmt --install all
```

run

### Branch Name

https://github.com/nasa/nos3/tree/dev

### Reproduction steps

```bash
sudo apt install binfmt

docker run --rm --privileged tonistiigi/binfmt --install all

docker run --rm --privileged tonistiigi/binfmt --uninstall qemu-x86_64
docker run --rm --privileged tonistiigi/binfmt --install all

run
```

### Screenshots

```bash
none
```

### Logs

```bash

```

### OS

Linux (Direct Clone)

jlucas9, 2025-05-28:
@byt3n33dl3 are you attempting to run on a Mac? I think the instructions you provided are from issue #706, which have no been vetted.

zlynch2, 2025-08-21:
Closing since not active since May, if issue still persists please create a new issue.

closed 2025-08-21 by zlynch2