nasa/nos3 #727: 🐛 [BUG] - CFDP Fails on Large Files on Older Laptops

msuder, 2025-06-05:
### Description

Older laptops are the old JSTAR machines that the interns are using.

Tried "truncating" files to various sizes.  900K file works ok to CFDP, 1M or 1000K file fails.

Note that on new EUSO laptops 1M file succeeds and so does a "truncated" 2M file.

### Branch Name

dev

### Reproduction steps

```bash
1.  Run through nominal scenario, manual CFDP steps, trying with dummy files of various sizes.
```

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

### Logs

```bash

```

### OS

Linux (Shared Folders)

Will-Matheny, 2025-06-05:
I timed the issues and I think it is because the download times out at 180 seconds.

jlucas9, 2025-06-06:
@Will-Matheny, can you try increasing that to confirm that resolves this issue? Whatever that number is that works on that old equipment we should update to be in NOS3. Thanks for your help!

jlucas9, 2025-06-06:
Seems to be an issue with https://nos3.readthedocs.io/en/latest/Scenario_Nominal_Ops.html

Was able to confirm:
[image]

In that script @Will-Matheny can you update line 28 to be 300 seconds and see if that works for you? If so feel free to make a branch, commit the fix, and submit a PR and we'll get this fixed!

PR #738 (2025-06-17): [nasa/nos3#727] Fixed Nom ops scenario with class 2 and timeout
Summary of PR goals.
During nominal operations scenario there is an error on older machines that fails the script. Fixed this by changing to Class 2 and doubled wait check time.

How to test?
* Run through nominal ops scenario.

Submodule PRs and actions prior to closing this:
* https://github.com/nasa-itc/gsw-cosmos/pull/37

Closes #727 .