nasa/nos3 #743: 🐛 [BUG] - File uplink failures

jechiasson, 2025-06-16:
### Description

Files fail to uplink via Cosmos to cFS.  Using sc-mission-config.xml. Tried multiple times with the same results.

### Branch Name

dev (e5e78a4a00f60233ceaec63683c24f2fbfb3d3da)

### Reproduction steps

```bash
Using cFS and cosmos with the sc-mission-config.xml configuration, uplink /tmp/nos3/uplink/tmp0.so and observe the FSW log output.
```

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

### Logs

```bash
Cosmos Command Sender:
cmd("CFDP SEND_FILE with CLASS 2, DEST_ID '24', SRCFILENAME '/tmp/nos3/uplink/tmp0.so', DSTFILENAME '/data/tmp0.s0'")

Cosmos Log:

2025/06/16 20:42:43.102  INFO: cmd("CFDP SEND_FILE with CLASS 2, DEST_ID '24', SRCFILENAME '/tmp/nos3/uplink/tmp0.so', DSTFILENAME '/data/tmp0.so'")
2025/06/16 20:42:43.102  INFO: UPLINK Class #2 transaction {21=>1} has started.
2025/06/16 20:42:43.103  INFO: Started processing file "/tmp/nos3/uplink/tmp0.so"
2025/06/16 20:42:43.104  INFO: Metadata sent for transaction {21=>1}
2025/06/16 20:42:43.271  INFO: Done processing file "/tmp/nos3/uplink/tmp0.so". Time elapsed: 0.169495195 seconds.
2025/06/16 20:42:43.276  INFO: Log File Opened : /home/johnchiasson/nos3-current/gsw/cosmos/outputs/logs/2025_06_16_20_42_43_cmd.bin
2025/06/16 20:42:58.410  INFO: EOF sent for transaction {21=>1}

FSW logs:

MGR: TimeTics = 1000000000022300 
EVS Port1 EVS Port1 42/1/CF 40: CF: md received for source: /tmp/nos3/uplink/tmp0.so, dest: /data/tmp0.so
MGR: TimeTics = 1000000000023300 
MGR: TimeTics = 1000000000024300 
MGR: TimeTics = 1000000000025300 
MGR: TimeTics = 1000000000026300 
EVS Port1 EVS Port1 42/1/CF 73: CF R2(21:1): ack limit reached, no fin-ack
MGR: TimeTics = 1000000000027300 
EVS Port1 EVS Port1 42/1/CF 63: CF: unhandled file directive code 0x04 in idle state
EVS Port1 EVS Port1 42/1/CF 63: CF: unhandled file directive code 0x04 in idle state
MGR: TimeTics = 1000000000028300 
EVS Port1 EVS Port1 42/1/CF 63: CF: unhandled file directive code 0x04 in idle state
EVS Port1 EVS Port1 42/1/CF 63: CF: unhandled file directive code 0x04 in idle state
MGR: TimeTics = 1000000000029300 
EVS Port1 EVS Port1 42/1/CF 63: CF: unhandled file directive code 0x04 in idle state
EVS Port1 EVS Port1 42/1/CF 63: CF: unhandled file directive code 0x04 in idle state
MGR: TimeTics = 1000000000030300 
EVS Port1 EVS Port1 42/1/CF 63: CF: unhandled file directive code 0x04 in idle state
EVS Port1 EVS Port1 42/1/CF 63: CF: unhandled file directive code 0x04 in idle state
MGR: TimeTics = 1000000000031300 
EVS Port1 EVS Port1 42/1/CF 63: CF: unhandled file directive code 0x04 in idle state
EVS Port1 EVS Port1 42/1/CF 63: CF: unhandled file directive code 0x04 in idle state
MGR: TimeTics = 1000000000032300 
EVS Port1 EVS Port1 42/1/CF 63: CF: unhandled file directive code 0x04 in idle state
EVS Port1 EVS Port1 42/1/CF 63: CF: unhandled file directive code 0x04 in idle state
MGR: TimeTics = 1000000000033300 
EVS Port1 EVS Port1 42/1/CF 63: CF: unhandled file directive code 0x04 in idle state
EVS Port1 EVS Port1 42/1/CF 63: CF: unhandled file directive code 0x04 in idle state
MGR: TimeTics = 1000000000034300 
EVS Port1 EVS Port1 42/1/CF 63: CF: unhandled file directive code 0x04 in idle state
EVS Port1 EVS Port1 42/1/CF 63: CF: unhandled file directive code 0x04 in idle state
MGR: TimeTics = 1000000000035300 
EVS Port1 EVS Port1 42/1/CF 63: CF: unhandled file directive code 0x04 in idle state
EVS Port1 EVS Port1 42/1/CF 63: CF: unhandled file directive code 0x04 in idle state
MGR: TimeTics = 1000000000036300 
EVS Port1 EVS Port1 42/1/CF 63: CF: unhandled file directive code 0x04 in idle state
MGR: TimeTics = 1000000000037300 

Cosmos Logs:

2025/06/16 20:45:08.805  INFO: Transaction {21=>1} has detected an error. Error is: EOF_LIMIT. Finishing transaction.
2025/06/16 20:45:08.805  INFO: Finishing transaction {21=>1}. Time elapsed: 145.703703085 seconds
```

### OS

Linux (Direct Clone)

jechiasson, 2025-06-18:
Good news - I pulled fresh dev and it's working now.  Bad news - I don't know what I did different.  I followed the scenarios both times.

jechiasson, 2025-06-20:
Now seeing issues downlinking dummy.txt:
EVS Port1 EVS Port1 42/1/CF 101: CF S1(24:1): received invalid nak pdu
EVS Port1 EVS Port1 42/1/CF 50: CF: nak pdu too short: 133 bytes received
EVS Port1 EVS Port1 42/1/CF 101: CF S1(24:1): received invalid nak pdu
EVS Port1 EVS Port1 42/1/CF 50: CF: nak pdu too short: 133 bytes received
EVS Port1 EVS Port1 42/1/CF 101: CF S1(24:1): received invalid nak pdu
EVS Port1 EVS Port1 42/1/CF 50: CF: nak pdu too short: 133 bytes received
EVS Port1 EVS Port1 42/1/CF 101: CF S1(24:1): received invalid nak pdu
EVS Port1 EVS Port1 42/1/CF 50: CF: nak pdu too short: 133 bytes received
EVS Port1 EVS Port1 42/1/CF 101: CF S1(24:1): received invalid nak pdu

Repeats indefinitely. There doesn't seem to be a way to cancel the operation.

jechiasson, 2025-06-26:
Thanks  @jlucas9, I appreciate the help.  I was able to move past this and get a build working on our sbc (and in docker). I had to create my own 'radio_sim' container that didn't depend on any of the engine or sims libs. There was a challenge getting the arm8 cryptolib, gpg-error, and socketcan libs from the container onto the sbc but once there cFS ran fine.

I've spent quite some time uplinking using Cosmos 4.5.0 and have learned the CF app doesn't resume a transfer if the link drops (LOS situation). I've got a proof of concept script working that chunks up the file and can resume when the link is up (AOS). It's not very robust, and I'm wondering if yamcs wouldn't be a better choice. Unfortunately we're running out of time (8 weeks left).  My bigger concern is understanding what to strip out of the remainder from the NOS3 default mission and what kind of tuning needs to be done.  There seems to be quite a bit of telemetry by default, and it'll need to be reduced to minimal status info.  Any advice on the shortest path?

jechiasson, 2025-06-27:
@jlucas9 , The issue is back and seems to be related to interrupted transferrs.  I've tried CF ABANDON, but there seems to be a packet size issue:

Cosmos:
2025/06/27 15:47:13.087  INFO: cmd("CFDP SEND_FILE with CLASS 2, DEST_ID '24', SRCFILENAME '/data/dummy.txt', DSTFILENAME '/data/dummy.txt'")
2025/06/27 15:47:13.087  INFO: UPLINK Class #2 transaction {21=>1} has started.
2025/06/27 15:51:54.587  INFO: cmd("CFS CF_ABANDON with TRANSACTION '0'")
2025/06/27 15:52:17.228  INFO: cmd("CFS CF_ABANDON with TRANSACTION '0.21_1'")
2025/06/27 15:52:45.197  INFO: cmd("CFS CF_ABANDON with TRANSACTION '21=>1'")

cFS:
EVS Port1 EVS Port1 42/1/CF 151: CF: invalid ground command length for command 0x09, expected 20 got 72
EVS Port1 EVS Port1 42/1/CF 151: CF: invalid ground command length for command 0x09, expected 20 got 72
EVS Port1 EVS Port1 42/1/CF 151: CF: invalid ground command length for command 0x09, expected 20 got 72

Donnie-Ice, 2025-06-27:
Try changing the `CCSDS_LENGTH` line in the following picture from 65 to 13 for the abandon command:

[image]

jechiasson, 2025-06-29:
Anything I can test on my end to help? We were counting on encrypted comms, so we are motivated. How about workarounds using other tooling?

Donnie-Ice, 2025-07-14:
CFDP now fixed in dev

closed 2025-07-14 by Donnie-Ice