nasa/nos3 #759: 🐛 [BUG] - Cannot see 42Sim frames

afatihsaruhan, 2025-07-12:
### Description

I can't see the 3d satellite and orbit models. 42Sim UIs not working on my VM.

### Reproduction steps

```bash
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
```

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

### Logs

```bash

```

### OS

Windows (Shared Folders), Linux (Direct Clone)

DitoluT, 2025-07-15:
Same here, and I have latest version of 42.

Donnie-Ice, 2025-07-15:
Are you using the vagrant install method with shared folders or a direct clone? Seems like we have a few people experiencing this.
@DitoluT @afatihsaruhan

Donnie-Ice, 2025-07-23:
We are working on a fix for our 1.7.3 release candidate. We are hoping to get the release out by the end of the week. Sorry for the issues!
@DitoluT @afatihsaruhan

zlynch2, 2025-07-31:
@DitoluT @afatihsaruhan 
we have just released 1.7.3 to our main branch, hopefully this solves some of your issues. I apologize for the inconvenience!