nasa/nos3 #768: amd64 issue with os_posix_time

zlynch2, 2025-07-25:
### Description

OS_Posix_TaskAPI_Impl_Init():394:Could not setschedparam in main thread: Operation not permitted (1)

this was noted in a rhel environment

### Reproduction steps

```bash
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
```

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

### Logs

```bash

```

### OS

Mac (Shared Folders)

haisamido, 2025-07-28:
@zlynch2 this may be related to https://github.com/nasa/nos3/issues/229

zlynch2, 2025-07-31:
It was found that Yamcs still works as long as the com interfaces are reset in the gui on startup. This is partly due to the preprocessor in java failing on boot up. However, yamcs still works with the os_posix time issus listed above and this is maybe due to permissions in Rocky. Still looking into it but noting how yamcs is working as it currently stands.